multiwii/multiwii-firmware #16: issues with mag and baro when i2c gps enabled

estechnical, 2015-08-27:
On multiwii v2.5 board with 328p if I enable i2c gps and msp wp command set the magnetometer and barometer update erratically and show odd values. Everything works fine with i2c gps disabled.

Fvk1960, 2015-08-27:
Are you able tot check each device adress is not overlapping!

estechnical, 2015-08-27:
Hi, thanks for quick reply, is there an obvious place to look for the sensor I2C addresses in the code or online?

Ed

Fvk1960, 2015-08-27:
In the software under sensors lookup the type of your baro and mag there in the code i2c adres is present. Now you have to find the code for the GPS. The adres on your GPS device should match.

estechnical, 2015-08-27:
A bit more info, I'm using tI2C_GPS_Nav 2.2 and the issue seems to appear once a GPS fix has been acquired. As soon as this happens I get garbage readings or none at all on mag and baro.

estechnical, 2015-08-27:
Also, I am using the default I2C addresses for the I2C GPS nav board firmware and have #define HK_MultiWii_SE_V2 in my config.h

estechnical, 2015-08-27:
Hi,

I have looked for the I2C addresses of the GPS nav board (default of
0x20 & matching in the corresponding nav board firmware) and the
MAG_ADDRESS 0x1E, BMP085_ADDRESS 0x77, MPU6050_ADDRESS 0x68 and I have
nothing else on the I2C bus.

It seems like the main 328 is unstable when I have the #define I2C_GPS
enabled.

Running everything from my bench supply and testing things through I
noticed a couple of unexpected resets when I powered on my TX. Nothing
like this happens with the I2C_GPS disabled, even if I leave it attached
to the I2C bus.

Any other ideas?

Many thanks,
Ed

Fvk1960, 2015-08-27:
Sorry no idea but you could try different multiwii versions from 2.1 till 2.4
Or check multiwii forum on i2c GPS
Good luck.

Fvk1960, 2015-08-27:
There is an topic on the forum but does not help very much but same problem appearance .
Topic connecting elements.all sensors gone Mad een adf GPS.
Secondly are both i2c on GPS and fc same voltage ie 5v versus 3.3v?
Thans all i can find on this.

estechnical, 2015-08-27:
Thanks for more info...

I have checked everything is on the same voltage I2c bus.

I've managed to narrow it down a bit... the problem occurs only when 5
or more satellites are in the gps data! As soon as this happens the mag
and baro data is useless but the accel and gyro data remains ok.

The cycle time becomes hugely erratic when the GPS fix is >5 satellites
too...

At least this gives something to look for.

Thanks,
Ed

estechnical, 2015-08-27:
I've found out that the 328p hasn't got enough memory for the following to all be enabled at once:
```
#define I2C_GPS
#define USE_MSP_WP
#define VBAT
#define POWERMETER_HARD
```
even though it compiles (with a warning about low memory).

As an experiment in finding out how close to the boundary this was in terms of memory, I changed:
```
unsigned char axis;
uint32_t dist;        //temp variable to store dist to copter
int32_t  dir;         //temp variable to store dir to copter
```
at the top of GPS_Compute() to be static and the problem with sensors existed immediately after flashing the board.

I reduced the memory of the power metering:
```

estechnical, 2015-09-10:
Hi,

I have tried a couple of flights since this issue appeared resolved, but I still don't trust the firmware that is compiled for the 328 if the i2c gps is enabled...

For some reason today my quad suddenly engaged GPS home mode when I was flying in Horizon mode (as if it thought it was outside the FENCE DISTANCE parameter) when hovering a few meters from me and I lost control and crashed. I was connected via bluetooth to the quad and EZgui announced the mode change.

I compiled the firmware from this github repo for atmega 328p using arduino 1.6.4.

My config file has QUADX, HK_MultiWii_SE_V2, failsafe, I2C gps, standard RC setup. Nothing much to note... it compiles and flies well in Horizon mode. I have now seen about 5 'incidents' which can only be described as something else taking over and driving for a while. I don't think that the firmware built by this combination of things is trustworthy.

Can anyone comment on this? Have you got this combination to work?

Thanks!

Hawke182, 2015-09-11:
Hi,

I have a similar Problem with an Serial GPS on serial2 with 57600 baud and UBLOX protocol.
My board is an CRIUS AIOP V2. with MultiWii 2.4.
As soon as the bord gets an GPS-fix the BARO and GPS-Home mode are switching on byself.
I can do nothing, just disarming.
Without GPS-Fix (in my room) everything is fine.

Matthias

estechnical, 2015-09-11:
Hi Matthias,

Very interesting... if you want to discuss this any more I also have a forum thread here: 
http://www.multiwii.com/forum/viewtopic.php?f=18&t=6709

Your issue (given the different hardware we have) doesn't sound like the same issue I was having, unless you're also running very low on memory.

Hawke182, 2015-09-11:
Thanks for the fast reply.
I have an ATMEGA 2560 on the board with 256 kByte Memory and just 34 kByte are used from the MultiWii 2.4. Plenty of free Memory.

Matthias

estechnical, 2015-09-11:
I was referring to the RAM, not program memory... 328p has 1k and 2560 is 8k. seems unlikely that it is memory... does your GPS indicate a stable lat & lon?

Hawke182, 2015-09-11:
I've just tested it again.
Lat lon and alt looks very stable 11 sat's.
The Baro and GPS-home are switching on only when armed.
Before and after arming it is off.

estechnical, 2015-09-11:
Have you tried clearing the eeprom and reloading the firmware + config?

estechnical, 2015-09-12:
From what I can tell, when the mode changes spontaneously it is due to something modifying the contents of `uint8_t  rcOptions[CHECKBOXITEMS];`

This is a serious bug if this is occurring - in my case I think this was happening because I was running out of ram and the heap/stack corruption that results.

In your case....???? I have no idea. Please can you share your config.h file so I can take a look?

Thanks,
Ed

Hawke182, 2015-09-13:
Everytime before I upload the Firmware I'm clearing the eeprom.
Yesterday I've downgraded to MultiWii 2.3_navi_baro_fix, same issue.
I've tried to commend the board-type and uncommented the single sensors instead of, but no change in the behavior.

Here is my config.h file:

#ifndef CONFIG_H_
#define CONFIG_H_

/*************************************************************************************************/
/****           CONFIGURABLE PARAMETERS                                                       ****/
/*************************************************************************************************/

/* this file consists of several sections
 * to create a working combination you must at least make your choices in section 1.
 * 1 - BASIC SETUP - you must select an option in every block.
 *      this assumes you have 4 channels connected to your board with standard ESCs and servos.
 * 2 - COPTER TYPE SPECIFIC OPTIONS - you likely want to check for options for your copter type
 * 3 - RC SYSTEM SETUP
 * 4 - ALTERNATE CPUs & BOARDS - if you have
 * 5 - ALTERNATE SETUP - select alternate RX (SBUS, PPM, etc.), alternate ESC-range, etc. here
 * 6 - OPTIONAL FEATURES - enable nice to have features here (FlightModes, LCD, telemetry, battery monitor etc.)
 * 7 - TUNING & DEVELOPER - if you know what you are doing; you have been warned
 *     - (ESCs calibration, Dynamic Motor/Prop Balancing, Diagnostics,Memory savings.....)
 * 8 - DEPRECATED - these features will be removed in some future release
 */

/* Notes:
 * 1. parameters marked with (*) in the comment are stored in eeprom and can be changed via serial monitor or LCD.
 * 2. parameters marked with (**) in the comment are stored in eeprom and can be changed via the GUI
 */

/*************************************************************************************************/
/*****************                                                                 ***************/
/****************  SECTION  1 - BASIC SETUP                                                *******/
/*****************                                                                 ***************/
/*************************************************************************************************/

  /**************************    The type of multicopter    ****************************/
    //#define GIMBAL
    //#define BI
    //#define TRI
    //#define QUADP
    #define QUADX
    //#define Y4
    //#define Y6
    //#define HEX6
    //#define HEX6X
    //#define HEX6H  // New Model
    //#define OCTOX8
    //#define OCTOFLATP
    //#define OCTOFLATX
    //#define FLYING_WING
    //#define VTAIL4
    //#define AIRPLANE
    //#define SINGLECOPTER
    //#define DUALCOPTER
    //#define HELI_120_CCPM
    //#define HELI_90_DEG

  /****************************    Motor minthrottle    *******************************/
    /* Set the minimum throttle command sent to the ESC (Electronic Speed Controller)
       This is the minimum value that allow motors to run at a idle speed  */
    //#define MINTHROTTLE 1300 // for Turnigy Plush ESCs 10A
    //#define MINTHROTTLE 1120 // for Super Simple ESCs 10A
    #define MINTHROTTLE 1064 // special ESC (simonk)
    //#define MINTHROTTLE 1050 // for brushed ESCs like ladybird
    //#define MINTHROTTLE 1150 // (*) (**)

  /****************************    Motor maxthrottle    *******************************/
[... 1098 lines cut ...]
    //#define COPTERTEST 1

/*************************************************************************************************/
/*****************                                                                 ***************/
/****************  SECTION  8 - DEPRECATED                                                 *******/
/*****************                                                                 ***************/
/*************************************************************************************************/

  /* these features will be removed in the unforseeable future. Do not build new products or
   * functionality based on such features. The default for all such features is OFF.
   */

  /**************************    WMP power pin     *******************************/
  //#define D12_POWER      // Use D12 on PROMINI to power sensors. Will disable servo[4] on D12
  /* disable use of the POWER PIN (allready done if the option RCAUXPIN12 is selected) */
  #define DISABLE_POWER_PIN

/*************************************************************************************************/
/****           END OF CONFIGURABLE PARAMETERS                                                ****/
/*************************************************************************************************/

#endif /* CONFIG_H_ */

Thanks for helping,
Matthias

estechnical, 2015-09-13:
Ok, thanks, I'll take a proper look soon...

Can you share a screenshot of your configuration in the GUI? Also, what is the radio setup like? You have a 3pos switch or similar as an Aux I assume?

Thanks,
Ed

closed 2016-01-09 by estechnical