multiwii/multiwii-firmware #27: Issue D4/Roll - NanoWii and QuadX

BenjaminJaume, 2019-08-28:
Good day,

When testing the MultiWii program, I don't have data coming from the pin D4 (ROLL). Although everything else works fine, I can read the PITCH, YAW, THROTTLE, etc.

My setup is:
- RC: Futaba T14SG
- Receiver: Futaba R7008SB
- Micro-controller: Arduino Nano (official)
- Drone: QuadX
- Channel 1 (ROLL) is connected to pin D4

Tests I've made:
- Downloading the program from the official source, not customised: working well, as expected
- Changing the copper wire: working well, as expected
- Plugging a servo on the Channel 1 (ROLL) receiver: working well, as expected
- Connecting the receiver's ROLL (channel 1) to pin D2 of the Arduino Nano: working well, as expected
- Connecting the receiver's ROLL (channel 1) to pin D4 of the Arduino Nano: not working
- Connecting the receiver's THROTTLE (channel 4) to pin D2 of the Arduino Nano: working well, as expected
- Connecting the receiver's THROTTLE (channel 4) to pin D4 of the Arduino Nano: not working
- Wiring an LED on pin D4 (ROLL) and run a blink program: working well, as expected

Do you guys have any ideas where that can come from?
Anyone is in touch with one of the programmers?

Thanks for your answers

Regards,
JorusTyron

mikhail-turicyn, 2019-10-01:
Maybe you should try sbus? Your receiver support it? just connect on 0(RX) pin and uncomment appropriate line in conf.h

BenjaminJaume, 2020-02-13:
I have actually found the solution. In the file `def.h`, I reversed the ROLL port with an unused one (AUX1 in my case). I ended up doing the following:

`
  #define THROTTLEPIN            2
  #define ROLLPIN                    7 // Switching with port AUX1PIN. Was 4 before.
  #define PITCHPIN                   5
  #define YAWPIN                     6
  #define AUX1PIN                    4 // Switching with port ROLLPIN. Was 7 before.
  #define AUX2PIN                    0 // optional PIN 8 or PIN 12
  #define AUX3PIN                    1 // unused 
  #define AUX4PIN                    3 // unused
`

closed 2020-02-13 by BenjaminJaume