multiwii/multiwii-firmware #29: I can't register on multiwi

tom-321, 2021-04-15:
I can't register on the multiwi forum and I can't get a reply via "Contact a Board Administrator".

When I try to register I just don't get a validation email.
I also already wrote an email to forum@multiwii.com.

tom-321, 2021-04-16:
Is there anybody out there?

treymarc, 2021-04-16:
yes, but here this is a GitHub mirror of the official SVN multiwii project .. and nothing else

can you try to contact Alex from the svn repository here -> https://code.google.com/archive/p/multiwii/

tom-321, 2021-04-17:
Hi thanks for the advice. 
I didn't know that this isn't part of the original project. 
And I didn't know who else to contact either. 

In the google code archive I found no email. 
Or is there any contact form ?