multiwii/multiwii-firmware #36: Motor 3 not getting throttle on command

ishanshah474, 2024-02-05:
Hello,

Setup - Used Arduino Nano, MPU6050 and BMP085 as sensors. Using the Multiwii Version 2.4. Configured the config.h file to use QuadX, enabled MPU6050 and BMP085 from independent sensors. Selected PPMSumReceiver on pin D2 using an ESP32. So basically, ESP32 receives input over Wi-fi as IP data packets and pipes it to pin D2 of the Arduino. 4 motors are on default pins (3, 9, 10 and 11). 

Now, I arm the drone, it gets armed, the 4 motors spin at minthrottle of 1064. So far so good. But when I send Throttle 25%, and roll, pitch and yaw to be 0%, all motors except the motor at pin D11 doesn't change speed. This is causing an imbalance and the drone to flip over. But when given random commands such as throttle 50%, yaw 100%, roll 0% and pitch 50%, it starts to give thrust (it changes speed) on D11 motor. But motors on D3 and D9 goes to no thrust (neutral speed, speed of 1064). Again, if we give random values, weird stuff keeps happening. 

Is there a solution to this? I already checked motor connections, calibrated ESC's and tried dynamic balancing. Please let me know if you figure out a problem. 

Thanks.

nik012003, 2024-02-05:
0% roll means the roll stick is fully set on the left, 50% would be middle (so neutral), and 100% means full right stick. Same logic applies for pitch.

ishanshah474, 2024-02-06:
> 0% roll means the roll stick is fully set on the left, 50% would be middle (so neutral), and 100% means full right stick. Same logic applies for pitch and yaw.

Thank you for your reply but I am sorry, but this is not what I meant by the problem. When I give 25% throttle, and 0% yaw, roll and pitch, the drone should lift up to a certain altitude and hold that, right? Since there is 0% of roll, pitch and yaw, it might not be stable, but it should just lift up. That means that all 4 motors should have equal or very close RPM. Motors 0, 1 and 2 have very similar RPM of around 1400 (on a throttle range of 1064-2000) but motor 3 has an RPM of 1064. That is causing and imbalance and the drone to flip upside down. As mentioned, I checked for motor connections, calibrated ESC's, made minimal adjustments to config.h but the problem still persists. Why is that only 1 motor does not get an equal RPM? Please let me know. Again, thank you.

nik012003, 2024-02-06:
If you give 0% (1000ns) yaw pitch and roll, you are basically asking the drone to bank full right, fully pitch back, and yaw left. If you want your drone to lift off upright, you should set those challnels to 50% (1500ns)

ishanshah474, 2024-02-07:
That is actually making sense to me now. I tried to do that. 50% for roll, pitch and yaw and 25% for throttle, now it has RPM in front right and rear left, but the other 2 motors still have neutral RPM. So, both CCW motors have almost equal RPM but the 2 CW motors have neutral RPM of 1064. Is this normal? Should I try taking off the drone with this? I've attached a screenshot of the RPM. Once again, thanks for your reply.
[image]

closed 2024-02-07 by ishanshah474