multiwii/multiwii-firmware #39: Change pin motor in multiwii 2.4

insonnecensore, 2024-05-02:
Good morning.
I need to change motor pin in multiwii 2.4.
I use Arduino Nano and NRF24L01 so I need pin 11 for MOSI.
Please help me to change pin number 11 into pint number 5 or 6.
Thanks.

zioCristia, 2024-05-05:
The motor pin order configuration is defined in the Output.cpp file.
For the Arduino Nano in particular is at line 20, you'll have to modify those lines accordingly.

For example:
```
#if defined(PROMINI)
  uint8_t PWM_PIN[8] = {9,10,5,3,6,5,A2,12};   //for a quad+: rear,right,left,front
#endif
```

Then of course, since the pin 5 is the pitch pin you'll have to modify the def.h file to change it (line 331 for Arduino Nano) with an unused one.

For example if you don't use the led pin:
```
#define PITCHPIN                   13
```

insonnecensore, 2024-05-06:
Thanks @zioCristia (I'm italian too).
Is it really so simple?
I'm going to use
#define PITCHPIN 8
because pin 8 is unused, and
#if defined(PROMINI)
  uint8_t PWM_PIN[8] = {9,10,5,3,6,5,A2,12};   //for a quad+: rear,right,left,front
#endif
That's all?
Nothing else for my **tricopter**?